ttlappalainen/NMEA2000 #412: N2k_CAN_int_pin

merco, 2024-07-12:
I'm using ESP32-WROOM-32D with mcp1525 board (8Mhz).
I'm using this declaration

#define N2k_CAN_INT_PIN 21

if I set N2k_CAN_int_pin I get this error 

[image]

[image]

ttlappalainen, 2024-07-12:
Do you use freetos or Arduino environment?

Why you do not use ESP32 internal CAN controller?

merco, 2024-07-12:
arduino.

I don't know, I can't get either one to work.
I would like to send (for example) env temp

I'm using these settings
#define USE_N2K_CAN 7
#define ESP32_CAN_TX_PIN GPIO_NUM_25  
#define ESP32_CAN_RX_PIN GPIO_NUM_26 

NMEA2000.SetN2kCANMsgBufSize(8);
  NMEA2000.SetN2kCANSendFrameBufSize(250);
  NMEA2000.SetN2kCANReceiveFrameBufSize(250);

  NMEA2000.SetProductInformation("1", // Manufacturer's Model serial code
                                 100, // Manufacturer's product code
                                 "My Sensor Module",  // Manufacturer's Model ID
                                 "1.0.2.25 (2019-07-07)",  // Manufacturer's Software version code
                                 "1.0.2.0 (2019-07-07)" // Manufacturer's Model version
                                );

  NMEA2000.SetDeviceInformation(112233, // Unique number. Use e.g. Serial number.
                                132, // Device function=Analog to NMEA 2000 Gateway. See codes on http://www.nmea.org/Assets/20120726%20nmea%202000%20class%20&%20function%20codes%20v%202.00.pdf
                                25, // Device class=Inter/Intranetwork Device. See codes on  http://www.nmea.org/Assets/20120726%20nmea%202000%20class%20&%20function%20codes%20v%202.00.pdf
                                2046 // Just choosen free from code list on http://www.nmea.org/Assets/20121020%20nmea%202000%20registration%20list.pdf
                               );
                               
  if (ReadStream!=ForwardStream) READ_STREAM.begin(115200);
  FORWARD_STREAM.begin(115200);
  NMEA2000.SetForwardStream(ForwardStream); 
 
  NMEA2000.SetMode(tNMEA2000::N2km_NodeOnly,25); 
  
  NMEA2000.SetForwardType(tNMEA2000::fwdt_Text); // Show bus data in clear text
    
  NMEA2000.ExtendTransmitMessages(TransmitMessages);
  if (ReadStream==ForwardStream) NMEA2000.SetForwardOwnMessages(false); // If streams are same, do not echo own messages.
  NMEA2000.EnableForward(false);
  NMEA2000.SetForwardOwnMessages(true);
  NMEA2000.Open();

and then in the loop I'm calling "SendN2kTempPressure"

void SendN2kTempPressure(void) {
  static unsigned long SlowDataUpdated = InitNextUpdate(SlowDataUpdatePeriod, TempSendOffset);
  tN2kMsg N2kMsg;

  double Temperature = 0;
  double BarometricPressure = 0;
  if ( IsTimeToUpdate(SlowDataUpdated) ) {
    SetNextUpdate(SlowDataUpdated, SlowDataUpdatePeriod);
    Temperature = 31.5;
    BarometricPressure = 1036;
    Serial.printf("Temperature: %3.1f °C - Barometric Pressure: %6.0f Pa\n", Temperature, BarometricPressure);
    Serial.println("");
    SetN2kPGN130310(N2kMsg, 0, N2kDoubleNA, Temperature, BarometricPressure);
    if (NMEA2000.SendMsg(N2kMsg)) {
      Serial.println("SENT!");
      N2kMsg.Print(&Serial);
    } else {
       Serial.println("!NO SENT!");
    }
  }
}

ttlappalainen, 2024-07-12:
If you use internal controller, have you properly connected transceiver to defined pins?

merco, 2024-07-12:
ESP32_CAN_TX_PIN GPIO_NUM_25
ESP32_CAN_RX_PIN GPIO_NUM_26

so pin 25 to transceiver TX
and 26 to transceiver RX

ttlappalainen, 2024-07-12:
Yes and depending of transceiver it may have other pins, which has to be propely connected. Read transceiver datasheets carefully.

merco, 2024-07-12:
this has only 3.3V and GND.

ttlappalainen, 2024-07-12:
Which transceiver?

merco, 2024-07-12:
SN65HVD230
For that reason I've done also a setup with mcp1525 . I'm waiting delivery of MCP2562.

ttlappalainen, 2024-07-12:
have you read the documentation? That chips has Rs pin "SN65HVD230 and SN65HVD231: Mode select pin: strong pull down to GND = high speed mode, strong
pull up to VCC = low power mode, 10kΩ to 100kΩ pull down to GND = slope control mode" So it has to be connected to GND directly or with resistor.

merco, 2024-07-12:
I can't believe it, I was throwing everything into the fire when the basic configuration with 2 SN65HVD230s started to work

[image]

merco, 2024-07-12:
A strange thing is happening.
Device A  is sending 130310 each 3 seconds and B is listening: 14580 : Pri:5 PGN:130310 Source:25 Dest:255 Len:8 Data:0,FF,FF,4E,C,A,0,FF

If I reset/turn off DeviceA, next time I turn on DeviceA I get this

[image]

until I reset also Device B

ttlappalainen, 2024-07-12:
If you have two devides, it is important that they have different NAME (read the document https://ttlappalainen.github.io/NMEA2000/md_7_glossary.html).

 ```
 NMEA2000.SetDeviceInformation(112233, // Unique number. Use e.g. Serial number.
                                132, // Device function=Analog to NMEA 2000 Gateway. See codes on http://www.nmea.org/Assets/20120726%20nmea%202000%20class%20&%20function%20codes%20v%202.00.pdf
                                25, // Device class=Inter/Intranetwork Device. See codes on  http://www.nmea.org/Assets/20120726%20nmea%202000%20class%20&%20function%20codes%20v%202.00.pdf
                                2046 // Just choosen free from code list on http://www.nmea.org/Assets/20121020%20nmea%202000%20registration%20list.pdf
                               );
```
So you must provide at least different unique number for each of them.