ttlappalainen/NMEA2000 #414: can't send ProductInformation

nett1s, 2024-07-16:
Hi, i making device that needs periodically send ProductInformation. I modified your Wind example, so i can send all another PGNs which i need, except ProductInformation. I use this function as you recommend using in other issues NMEA2000.SendProductInformation(); but nothing i see in Actisense listner. Other PGN's i see, this one no.
at the start my code:

const unsigned long TransmitMessages[] PROGMEM = {126996L, 0};

tN2kSyncScheduler ProductInformationScheduler(false, 5000, 1);

void OnN2kOpen() {
  ProductInformationScheduler.UpdateNextTime();
}

 Also I added in setup NMEA2000.SetN2kCANSendFrameBufSize(200); (Tryed with different values).  After this function i placing:

  NMEA2000.SetProductInformation( //My data...//); 
  NMEA2000.SetDeviceInformation( //My data...//); 
  NMEA2000.SetMode(tNMEA2000::N2km_ListenAndNode,22);
  NMEA2000.EnableForward(false);
  NMEA2000.ExtendTransmitMessages(TransmitMessages);
  NMEA2000.SetOnOpen(OnN2kOpen);
  NMEA2000.Open();
and in the loop as in wind example:
{
SendN2kProductInformation();
NMEA2000.ParseMessages();
}

void SendN2kProductInformation() {
  tN2kMsg N2kMsg;
  if ( ProductInformationScheduler.IsTime() ) {
    ProductInformationScheduler.UpdateNextTime();
    NMEA2000.SendProductInformation();
  }
}

ttlappalainen, 2024-07-28:
Why do you need to send product information? It will be automatically send by library on request.

I did not see errors in your code. You do not need to add 126996L to PGN lisn, since it is system message and will be automatically added. Which CAN controller and driver library you use?

nett1s, 2024-07-29:
I'm using Atmega1284p with MightyCore (maybe problem in this), MCP2515. It will not ack because if try to send by hand adding NMEA2000.SendProductInformation(), so after few messages will be tryed to send all hangs. I thinking that problem in MightyCore libs, so it just hangs from this big message. Maybe buffer overflawing and hanging microcontroller.

ttlappalainen, 2024-07-30:
You have to check how much you have memory remaining during run. You should not define big buffers for low memory devices. Read from document (https://ttlappalainen.github.io/NMEA2000/classt_n_m_e_a2000.html#a5e09e469c4126963e26802c4541117bb) about required buffer definition. You should also define receive buffer size - for NMEA2000_mcp default is 1.

nett1s, 2024-07-30:
I all this already tryed, no effect, it just hangs after few messages which i see in Actisense Listner. But if i remove sending Product Information, all works. It's very strange, it seems i will must to change microcontroller, because i can't find any possible dicision.

ttlappalainen, 2024-07-31:
Sounds buffer overrun.

Why not use ESP32 (not S-models) or very powerful Teensy4? Both are vell tested and used in certified products.