ttlappalainen/NMEA2000 #415: (Help) UDP by esp32

peppeve, 2024-07-20:
Is possible implement the UDP stream by esp32 of received messages? some software are able to read this UDP stream and use N2K messages.

ttlappalainen, 2024-07-21:
You may mix things a bit. NMEA2000 library core takes care of NMEA2000 (CAN) bus communication and necessary required tasks. On UDP stream you have just data coming, which means that you need some kind of format converter instead of full library. UDP stream format is not specified as standard. So you may have:
1. Actisense format. Then you can use ActisenseReader module to generate tN2kMsg and then use N2kMessages for parsing.
2. SeaSmart format. Then you ca use SeaSmart module to generate tN2kMsg and then use N2kMessages for parsing.
3. Bare frame format. Needs some logic to build messages to tN2kMsg and then use N2kMessages for parsing.
4. Some other format. Needs some logic to build messages to tN2kMsg and then use N2kMessages for parsing.

Note that UDP data can be in what ever format sender has decided to use. Even bare frame format can be ascii or binary.

peppeve, 2024-07-21:
Sorry Timo, i try to expain better.
Some professianal marine software are able to read actisense format nmea data for the UDP and not only for serial.
My doubt is how use. Something like this will be fine?

`
N2kMsg.SendInActisenseFormat(&UDP);
....
Udp.beginPacket(Udp.remoteIP(), Udp.remotePort());
NMEA2000.ParseMessages();
Udp.endPacket();
`

ttlappalainen, 2024-07-21:
You can just open stream and set it as forward stream. You do not need to use beginPacket/endPacket on loop.

That method sends all messages to the stream, which is a bit wasting of resources, if you do not want to analyze all messages. Other method is just handle each message and on message handler send only selected messages with tN2kMsg::SendInActisenseFormat
```
void HandleNMEA2000Msg(const tN2kMsg &N2kMsg) {
  switch ( N2kMsg.PGN ) {
    case 126992:
    case 127250:
    case 127488:
    case 127489:
    case 129026:
    case 129029:
      N2kMsg.SendInActisenseFormat(&udp);
      break;
  }
}

```
Remember to use AsyncUDP library to avoid blocking.

peppeve, 2024-07-21:
Thanks Timo. But how I can define in this case the IP address and port where send the udp packets?

ttlappalainen, 2024-07-21:
When you open UDP stream. I have tested only broadcast and just kept udp open all the time. If you need to use addressed, you have to test that can it be open all the time as far and if connection closes, then try to reopen. If you open/close it all the time, you have to test that it does not block loop and possibly move sending to an other thread. If you need to use other thread, you have to put received messages to some list and take care of data locking with mutexes.

peppeve, 2024-07-21:
My IP esp32 is 192.168.0.15 and I want to send the udp packages in all my internal network 192.168.0.255 at the 5000 port.

ttlappalainen, 2024-07-21:
Then you simply use tUDPBroadcastStream, open it at setup and either use forward stream or on message handler as I wrote above. 
[UDPBroadcastStream.zip](https://github.com/user-attachments/files/16324059/UDPBroadcastStream.zip)

peppeve, 2024-07-22:
Hi Timo, it work but seem to be some problems. the stream (text) seem to be wrong in the udp side, serial side it is fine. I attach two filese with log of both serial and udp.
[Serial.txt](https://github.com/user-attachments/files/16328945/Serial.txt)
[UDP 60001.log](https://github.com/user-attachments/files/16328946/UDP.60001.log)

ttlappalainen, 2024-07-22:
You have set: NMEA2000.SetForwardType(tNMEA2000::fwdt_Text);
Remove or comment that line.

peppeve, 2024-07-22:
Yes i Know, but i suspect the Actisense format is wrong too on UDP stream, if you check the two files in text format, serial stream is ok UDP stream is wrong

ttlappalainen, 2024-07-22:
Both are not in actisense format. They are both in text format, which is only for testing. If you use forwarding, then you have to use fwdt_Actisense format or remove line I mentioned. If you use messagehandler, then use N2kMsg.SendInActisenseFormat(&outputStream);

peppeve, 2024-07-22:
The UDP stream in actisense format seem to be very fine (respect to Text stream). I attach a log file.
Thank's for yours help
[UDP 60001.log](https://github.com/user-attachments/files/16329654/UDP.60001.log)