ttlappalainen/NMEA2000 #416: ESP32C3 Compilation Error

peppeve, 2024-07-25:
I get this error compiling the message sender example.

c:\Users\User\Documents\Arduino\libraries\NMEA2000_esp32\NMEA2000_esp32.cpp:33:10: fatal error: soc/dport_reg.h: No such file or directory
   33 | #include "soc/dport_reg.h"
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: exit status 1

The ESP32C3 supports a two-wire automotive interface (TWAI), which is compatible with ISO11898-1 classical frames CAN 2.0

jiauka, 2024-07-25:
NMEA2000_esp32 is not compatible with esp32c3, you may try 
https://github.com/jiauka/NMEA2000_esp32xx

peppeve, 2024-07-25:
#include <NMEA2000_esp32xx.h>

but i get same error.

fatal error: soc/dport_reg.h: No such file or directory
   33 | #include "soc/dport_reg.h"

jiauka, 2024-07-25:
You have to disable NMEA2000_ESP32

peppeve, 2024-07-25:
Already done, modified nmea2000_can.h

//#elif USE_N2K_CAN == USE_N2K_ESP32_CAN
//#include <NMEA2000_esp32.h>       // https://github.com/ttlappalainen/NMEA2000_esp32
//tNMEA2000 &NMEA2000=*(new tNMEA2000_esp32());

#elif USE_N2K_CAN == USE_N2K_ESP32_CAN 
#include <NMEA2000_esp32xx.h>     // https://github.com/ttlappalainen/NMEA2000_esp32 
tNMEA2000 &NMEA2000=*(new tNMEA2000_esp32xx()); 

but i get this error:

In file included from c:\Users\Motopesca\Documents\Arduino\libraries\NMEA2000\src/NMEA2000_CAN.h:281,
                 from C:\Users\Motopesca\Documents\Arduino\MessageSenderVedda\MessageSenderVedda.ino:34:
c:\Users\Motopesca\Documents\Arduino\libraries\NMEA2000_esp32xx/NMEA2000_esp32xx.h:75:3: error: 'twai_handle_t' does not name a type; did you mean 'intr_handle_t'?
   75 |   twai_handle_t  Handle;
      |   ^~~~~~~~~~~~~
      |   intr_handle_t

exit status 1

jiauka, 2024-07-25:
Which Arduino version are you using?

ttlappalainen, 2024-07-26:
You do not need to modify NMEA2000_CAN.h. Just do not include it. Instead define what you want to use and create NMEA2000 object.

```
...
#include "N2kMsg.h"
#include "NMEA2000.h"
#include <NMEA2000_esp32xx.h>
#include <N2kMessages.h>

tNMEA2000 &NMEA2000=*(new tNMEA2000_esp32xx());
...
```

Above does not help for you last error. It is related to Arduino version you use.

Note that NMEA2000_esp32xx.h does not use properly NMEA2000 library features, so it will not be added to NMEA2000_CAN.h, which is easy for beginners and for testing, if you use same code for different MCU:s

peppeve, 2024-07-26:
I'm using Arduino IDE 2.3.2

peppeve, 2024-07-31:
Any help will be appreciated

peppeve, 2024-07-31:
Solved with updated version of the NMEA2000_esp32xx library

closed 2024-07-31 by peppeve