ttlappalainen/NMEA2000 #419: N2K out but no N2K in.

dansonamission, 2024-08-03:
Hello

Have been using this library for a while now and have a few working projects using Adafruit CAN Feathers as well as our own RP2040 boards. Yesterday I tried to use a Adafruit QT Py with a Adafruit CAN BFF (MCP2515), after much trial and error the board will transmit NMEA 2000 ok, but it will not receive anything.

To test the CAN I loaded the CAN_BUS_SHIELD Receive_check example and can see all the CAN messages being received in the serial console.

Is there anything obvious to check that I might have missed as I've been staring at it for too long now?

Thanks

ttlappalainen, 2024-08-04:
Did you defined IRQ and receive buffer size properly?

dansonamission, 2024-08-04:
It’s an issue when we define the interrupt pin. With no pin defined the receive works ok.  It’s actually a MCP25625, same as the adafruit CAN feather which we have no issues with. 

We’re using default buffers, but have also been using the arduino gateway example which has buffers defined.

ttlappalainen, 2024-08-05:
So your problem is actually that interrupt with your board does not work. CAN_BUS_SHIELD Receive_check works, since it does not use interrupts. You can use also NMEA2000 library without interupt, but then you have to take care that you call tNMEA2000::ParseMessages faster than possible message burst. If I remember right MCP2515 has 8 frame receive buffer and CAN can have about 2 message/ms, so you should call it at least with 4 ms period to avoid frame loss. If you have some other libraries causing delays, you will most probably loose some important messages time to time. With interrupting frames will be buffered to larger receive buffer.

Even with interrupting system it is good to keep tNMEA2000::ParseMessages period low to have good timing. Delay between two tNMEA2000::ParseMessages calls should not exceed 200 ms even randomly.