ttlappalainen/NMEA2000 #420: MIssing PGN Electric Motor drive status 128002

chrisclaydon, 2024-08-07:
Hi there,
It would be great if you can implement this PGN. I have this list of the fields:

128002 Electric Drive Status (Rapid Update)
1 Inverter/Motor Controller (specifies which system the PGN message relates to.)
2 Active Motor Mode
3 Brake Mode
4 NMEA Reserved
5 Rotational Shaft Speed
6 Motor DC Voltage
7 Motor DC Current

Also found this information on the canboat website:
0x1F402: PGN 128002 - Electric Drive Status, Rapid Update
This PGN is used to provide the Electric Propulsion Drive System Status.

This PGN is not fully reverse engineered. Some aspects that are known to be missing/incorrect:

The length of one or more fields is likely incorrect
The resolution of one or more fields is likely incorrect
One or more of the lookup fields contain missing or incorrect values
The PGN has not been seen in any logfiles yet
The default transmission interval is not known
This single-frame PGN is 8 bytes long and contains 7 fields.

Field #	Field Name	Description	Unit	Type
1	Inverter/Motor Controller		
0 .. 253
8 bits unsigned [NUMBER]
2	Active Motor Mode		
0 .. 2
2 bits unsigned [NUMBER]
0 .. 2
2 bits unsigned [NUMBER]
4	Reserved			4 bits [RESERVED]
5	Rotational Shaft Speed		0.25 [rpm]
0 .. 16383.25
16 bits unsigned [NUMBER]
6	Motor DC Voltage		0.1 [V]
0 .. 6553.3
16 bits unsigned [NUMBER]
7	Motor DC Current		0.1 [A]
-3276.7 .. 3276.5
16 bits signed [NUMBER]

ttlappalainen, 2024-08-07:
it is not very common, so it does not need to be part of the library. If you have information for it, you can make own module for it under you own code.

ttlappalainen, 2024-08-09:
Not thousands of hours...