ttlappalainen/NMEA2000 #421: Multimedia Control

SlimSchip, 2024-08-08:
Hi,

I am about to do a little project  where I need to make a device that acts as a N2K multimedia interface. so I am able to control volume/source/prev/next/play/pauze etc.

Does anyone have any insights on the procedure to follow, I have found the PGN's but have no information on about how the  discovery works. ie 126996/8 replies etc.

Anyone who can help me implement this?

M.

ttlappalainen, 2024-08-08:
Library does automatically responces to the main PGNs like 60928, 126996/8 request.

FlaminMad, 2025-07-13:
Hi @SlimSchip,

How did you get on with this project?
I've got a Garmin UHD chartplotter which is NMEA2000 capable, and am currently exploring if I could build an interface to control an amp which it would recognise.

Thanks!