ttlappalainen/NMEA2000 #425: NMEA2000__mcp library "Type not declared error"

CD33-Jay, 2024-08-28:
Trying to use a MCP CAN module with an esp32 using PlatfromIO.  I've used your TemperatureMonitor.ino example to test.  I'm getting 
" error: 'MCP_16MHz' was not declared in this scope" and  "error: 'class MCP_CAN' has no member named 'setSPI';" error messages.

Here's my platform.ini file, the beginning of my TemperatureMonitor.cpp file (rest of the file not shown is same as example) and my compile error messages.

platformIO.ini
[temp_platformio.txt](https://github.com/user-attachments/files/16782078/temp_platformio.txt)

TemperatureMonitor.cpp
[temp_cpp.txt](https://github.com/user-attachments/files/16782083/temp_cpp.txt)

Compile errors
[temp_compile_error.txt](https://github.com/user-attachments/files/16782085/temp_compile_error.txt)

Am I doing something wrong?
Thanks,
Jay

CD33-Jay, 2024-08-28:
I found that I had the wrong mcp_can.h library.   Switched to CAN_BUS_SHIELD library and now have a new error:

> In file included from .pio/libdeps/esp32doit-devkit-v1/NMEA2000-library/src/NMEA2000_CAN.h:317,
                 from src/main.cpp:23:
.pio/libdeps/esp32doit-devkit-v1/NMEA2000_mcp/NMEA2000_mcp.h:45:11: error: cannot declare field 'tNMEA2000_mcp::N2kCAN' to be of abstract type 'MCP_CAN'
   MCP_CAN N2kCAN;
           ^~~~~~
In file included from .pio/libdeps/esp32doit-devkit-v1/NMEA2000-library/src/NMEA2000_CAN.h:296,
                 from src/main.cpp:23:
.pio/libdeps/esp32doit-devkit-v1/CAN_BUS_Shield/src/mcp_can.h:55:7: note:   because the following virtual functions are pure within 'MCP_CAN':
 class MCP_CAN
       ^~~~~~~
.pio/libdeps/esp32doit-devkit-v1/CAN_BUS_Shield/src/mcp_can.h:58:18: note:      'virtual void MCP_CAN::enableTxInterrupt(bool)'
     virtual void enableTxInterrupt(bool enable = true) = 0;                             // enable transmit interrupt
                  ^~~~~~~~~~~~~~~~~
....... (and many more of these)

Appreciate any help.  Thanks
Jay

closed 2024-08-28 by CD33-Jay

ttlappalainen, 2024-08-28:
Just curious. Why you do not use ESP32 internal CAN and just add suitale transceiver? mcp_can should be ok, but it has not been used in certified devices as far as I know. ESP32 internal CAN has been instead passed certification tests.

CD33-Jay, 2024-08-28:
Yes I'm aware of the internal CAN capability of ESP32.   My real application is similiar to espberry (issue #343).  I'm trying to implement a 2 CAN bus solution you sugggested in #343.  My application is to correct the apparent wind angle data from a masthead wind instrument mounted on a rotating mast on my trimaran.  The wind PGN is read into  an esp32 and corrected based on mast rotation angle measured with a Honeywell sensor at the base of the mast.  This corrected wind data is sent out on the NMEA2000 bus.   The chart plotter(MFD) sees the two wind PGN's, one uncorrected and one corrected.  When the corrected wind PGN is selected as default on the MFD, the MFD turns the masthead wind instrument off!  Therefore the esp32 does not recieve the uncorrected wind data.   The solution is to seperate the two CAN buses to prevent the MFD from receiving two wind PGNs.

I only decided to try a simple example problem from the library to simplify my issue post to something easily replicable.

Thanks for the wonderful NMEA2000 library.  Looking at the code, your library is impressive.  This would not be possible for me to implement with it.
Thanks again!
Jay