ttlappalainen/NMEA2000 #427: SetN2kEngineDynamicParam does not support negative numbers for oil pressure

donmarlin, 2024-09-08:
I have a question about "SetN2kEngineDynamicParam" (PGN 127489). 
Seems if I send a negative oil pressure out using N2K, it is not represented correctly on the bus. 
The function prototype declares this as a double so I assumed signed values would be handled ok.

For example, sending -7000pa (~-1 psi) out using "SetN2kEngineDynamicParam",  the value on the bus is actually showing as 6553400. 
I am reading back using the "ParseN2kEngineDynamicParam" on a separate device as well as the Actisense NMEA Reader app.

I will work around since negative oil pressure is abnormal and I can clamp to zero. I was only wondering if it was something with my environment or my understanding. Thanks in advance

ttlappalainen, 2024-09-08:
You can not set negative values for any pressure on engine dynamic PGN. If you set, library changes them to out of range value 6553400 Pa

donmarlin, 2024-09-08:
Thanks for the reply, appreciate the timely feedback.

closed 2024-09-08 by donmarlin