ttlappalainen/NMEA2000 #429: N2K connection with ISO1050 and separate power supply, no power CAN-side

terolindberg, 2024-09-20:
Hi,

First, amazing job with the lib, thank you!

Maybe an issue or my lack of grasping the handling properly.
I have an esp32 with ISO1050 connected to NMEA2000, the esp is powered separately from the bus and is "always-on". 
If I have the esp powered up and then power the CAN bus, the NMEA messages won't appear in the bus. If I then restart the esp while the bus is powered up, the messages start flowing fine from the ESP to the other devices. Esp at the moment only transmitting, not listening to bus. 
I suspected this has something to do with ISO1050 "Undervoltage Lockout and Fail-Safe" but it should clear quickly after power is on on the bus.
Is this something to do with the library, or is there some trick I'd need to do with ISO1050 or the lib?

Thanks for the advice in advance!

virtuvas, 2024-09-20:
Timo,

I've come across that a few times and I'm about to program a LilyGO T display with an embedded esp32 chipset as a tiny touchscreen controller for some N2K functions.
Wouldn't mind having a look at that code if you don't mind!

cheers

V.

ttlappalainen, 2024-09-23:
Do you mean my autoreset functionality?

virtuvas, 2024-09-24:
> Do you mean my autoreset functionality?

yes, not in a hurry though, need to get into winter programming mode first!

V.

ttlappalainen, 2024-09-24:
No idea. I have several devices and the one failing has same information than one other, which does not fail.