ttlappalainen/NMEA2000 #431: No CanGetFrame when connecting ActisenseListenerSender.ino to a different computer

MartinVerges, 2024-09-30:
Hello,

I have a very strange issue that I try to debug but somehow can't figure out why. I use the `ActisenseListenerSender.ino` sketch and upload it on a ESP32 with a ISO1050DUB CAN-Transceiver. I usually use a Linux laptop with Visual Studio Code and a small platformio project to manage it. That works fine and I can see the output on the serial console, so perfect. I therefore assume, that the hardware is ok.
Now I wanted to connect this ready to use device to a Windows 10 panel PC but the serial output shows no sign of CAN data. Everything else works fine, so the CP210x driver detects the ESP32 and I can open the Serial console using the same setup (visual studio code, same platformio project,...) but besides some ESP32 boot messages and CAN device ready message, nothing pops up. I put both devices next to each other, so I only move the USB cable from my Laptop to the Tablet PC and don't move or touch the ESP32 or it's CAN connection. Just the USB and the output is gone.

To debug it, I tried to put some Serial.prints into your code and end up in `ParseMessages` line `while (FramesRead<MaxReadFramesOnParse && CANGetFrame(canId,len,buf) ) { `. The part `FramesRead<MaxReadFramesOnParse` is `true`, but `CANGetFrame` always returns `false` when it is connected to the Windows 10 and true when it's on my Linux.

How and why does the USB connected Host affect the CAN Frame part of the ESP32? By chance, do you have any idea what that could be?

Thanks in advance
Martin

ttlappalainen, 2024-09-30:
Host will not affect to library. CANGetFrame(canId,len,buf) can not always be true. It will be true, when new frame has been catched. Frames will be read from CAN controller to receive buffer with interrupts. If it never returns true, it means that no CAN frames as been received by CAN controller so CAN communication does not work.

My guess is that you have problem with system powering and there is some difference between laptops.
- You must power NMEA2000 bus with separate PSU
- You must have ISO1050 CAN side powered with separate DC-DC converter from CAN side and ESP32 side and ESP32 itself with own PSU. You can also power all from CAN side or ESP32 side, but then you must use isolated DC-DC converter and have to take care not meshing up with USB powering and handshaking signals.
- One common problem has been meshing up with NMEA2000 bus powering. Just few weeks ago after several checks someone had problem and could not solve it. I asked to do one more check and he found that while disconnecting one device he also turned off bus power (or something - do not exactly remember how it went).

MartinVerges, 2024-09-30:
The can bus of my boat says active during the time and is powered
independent of this ActisenseListener.
The esp32 however is powered by the host right now, I can try a seperate
PSU for that, let's see.

It's a busy boat bus and there's a lot of information coming in,
`CANGetFrame` is not always true, but basically nonstop due to messages.

Again, I touch nothing else than the USB cable. Everything works on the
boat and is powered up. I disconnect the USB cable from my Linux Laptop
where I can see the correct serial output and connect it 10cm away into
this windows panel pc and it shows only the boot message. I pull it again,
plug it back into the laptop and the serial output shows thousands of
frames. Nothing else touched.

Would it help if I record a video of that? It's super strange as I tried
another Windows 11 laptop and it works fine with it. It's just not working
on this one panel pc where I needed it for. That's why I somehow believe
that the hosts is affecting it somehow.
Best regards
Martin Verges
+49 174 9335695 (Mobile, Telegram, WhatsApp,..)

MartinVerges, 2024-09-30:
> You can also add to your loop some message every 1 sec to see that it is really running and not blocked to e.g., programming state.

That's what I did. I added `serial.print` in `loop` then in `tNMEA2000::ParseMessages()` until I found out that it never goes into the `while` loop as `CANGetFrame(canId,len,buf)` always (yes, always) returns false on the panel PC. So I'm very sure it does the loop, prints out that `FramesRead<MaxReadFramesOnParse` results in true and never prints out inside the while loop. I even tried to dig into the `xQueueReceive` within the `CANGetFrame` but unfortunately that`s more than my knowledge allows me to debug. From my point of view it looks like as if the Queue doesn't get populated while it's USB connected to the panel PC, super strange.

I will try to connect it to an external power source to make the ESP power independent from the USB and hope this helps. Will update here later today after trying that out.

ttlappalainen, 2024-09-30:
As I mentioned, it never gets any frame from bus and thats why it returns false. It means that CAN controller can not get any message from the bus. Connecting to different USB does not change code.

Do you have schematic of your ISO1050 connection?

MartinVerges, 2024-09-30:
[image]
[image]
[image]
[image]

I agree that the code on the ESP32 does not change because of plugging in the USB somewhere else, but as I don't even touch the NMEA2k side and my display that is connected in parallel shows the information of the bus, I'm pretty sure (like 99,99%) that there are frames in the CAN that are signaling towards the ESP32. That's why I opened the issue as it's so weird.

Serial Log on my Linux Laptop with debug enabled looks fine.
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
[    18][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[    35][V][esp32-hal-uart.c:330] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(3) txPin(1)
[    44][V][esp32-hal-uart.c:416] uartBegin(): UART0 not installed. Starting installation
[    55][V][esp32-hal-uart.c:463] uartBegin(): UART0 initialization done.
---- SETUP START ----
Initialize frame buffer. Size: 4, address:1073450172
Init devices
CAN device ready
---- Start loop() ----
318: Received frame, can ID:167248536 len:8 data:FF,FC,63,27,2,0,FF,FF
318: Use msg slot: 0
318: Single frame=129026
319:  - Non system message, MsgIndex: 0
329 : Pri:2 PGN:129026 Source:152 Dest:255 Len:8 Data:FF,FC,63,27,2,0,FF,FF
329:  - Free message, MsgIndex: 0
340: Received frame, can ID:167248280 len:8 data:10,EE,9F,8,C0,88,B8,DB
... lots of these lines
```
again, only unplugging USB from the Laptop and plugging it into the Panel PC gives:
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
[    18][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[    35][V][esp32-hal-uart.c:330] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(3) txPin(1)
[    44][V][esp32-hal-uart.c:416] uartBegin(): UART0 not installed. Starting installation
[    55][V][esp32-hal-uart.c:463] uartBegin(): UART0 initialization done.
---- SETUP START ----
Initialize frame buffer. Size: 4, address:1073450172
Init devices
CAN device ready
---- Start loop() ----
```
no other message appears as long as I don't add more `serial.print`.

I tried the external PSU, unfortunately without success. Same behavior.
Any idea how to debug it further?

ttlappalainen, 2024-09-30:
One problem you have are R14 and R15. Remember that bus resistance it 60 ohm and now you drive it trough 100 ohm. You should drive it without resistors.

MartinVerges, 2024-09-30:
Thanks, it's not my design but I simple removed them and bridged R14 and R15 as I agree this could be an issue. Unfortunately it did not change this super strange behavior.

So I added some serial.print again and tried:

```
--- forcing DTR inactive
--- forcing RTS inactive
--- Terminal on /dev/ttyUSB0 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
[    18][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[    35][V][esp32-hal-uart.c:330] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(3) txPin(1)
[    44][V][esp32-hal-uart.c:416] uartBegin(): UART0 not installed. Starting installation
[    55][V][esp32-hal-uart.c:463] uartBegin(): UART0 initialization done.
Firmware build at: 09:54:11 on Sep 30 2024
---- SETUP START ----
Initialize frame buffer. Size: 4, address:1073450172
Init devices
CAN device ready
---- Start loop() ----

.xQueueReceive = false
.xQueueReceive = false
.... 104 times ...
.xQueueReceive = false
.xQueueReceive = false
.X291: Received frame, can ID:166792620 len:8 data:FC,F8,FF,7F,58,7,FF,FF
292: Use msg slot: 0
292: Single frame=127245
292:  - Non system message, MsgIndex: 0
303 : Pri:2 PGN:127245 Source:172 Dest:255 Len:8 Data:FC,F8,FF,7F,58,7,FF,FF
303:  - Free message, MsgIndex: 0
313: Received frame, can ID:167052067 len:8 data:FF,0,0,FF,FF,0,FF,FF
314: Use msg slot: 0
314: Single frame=128259
324:  - Non system message, MsgIndex: 0
324 : Pri:2 PGN:128259 Source:35 Dest:255 Len:8 Data:FF,0,0,FF,FF,0,FF,FF
335:  - Free message, MsgIndex: 0
335: Received frame, can ID:166793932 len:8 data:FF,19,38,FF,7F,FF,7F,FD
336: Use msg slot: 0
346: Single frame=127250
346:  - Non system message, MsgIndex: 0
346 : Pri:2 PGN:127250 Source:204 Dest:255 Len:8 Data:FF,19,38,FF,7F,FF,7F,FD
357:  - Free message, MsgIndex: 0
357: Received frame, can ID:166794188 len:8 data:FF,37,35,1,0,FF,FF,FF
367: Use msg slot: 0
... many many more
703: Use msg slot: 1
703: First frame=126208

.X703: Received frame, can ID:233688236 len:8 data:C1,1,3B,7,3,4,4,6C
714: New frame=126208 frame=C1
... many many more
```

please take a look at the firmware build date that I added as well, again not changing the NMEA2k connection, not touching the external PSU, just moving the USB cable from the Laptop to the tablet PC results in:

```

--- forcing DTR active
--- forcing RTS active
--- Terminal on COM4 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
[    18][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[    35][V][esp32-hal-uart.c:330] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(3) txPin(1)
[    44][V][esp32-hal-uart.c:416] uartBegin(): UART0 not installed. Starting installation
[    55][V][esp32-hal-uart.c:463] uartBegin(): UART0 initialization done.
Firmware build at: 09:54:11 on Sep 30 2024
---- SETUP START ----
Initialize frame buffer. Size: 4, address:1073450172
Init devices
CAN device ready
---- Start loop() ----

.xQueueReceive = false
.xQueueReceive = false
.... 104 times ... (interestingly the exact same amount)
.xQueueReceive = false
.xQueueReceive = false
.XxQueueReceive = false
.XxQueueReceive = false
... forever
```

I added:
`Serial.print("\n.");` as the first call inside `loop()`
`if(FramesRead<MaxReadFramesOnParse) Serial.print("X");` just before the `while` loop inside `tNMEA2000::ParseMessages()`
`else Serial.print("xQueueReceive = false");` inside `tNMEA2000_esp32::CANGetFrame`.

Any other Ideas what I can try or add to debug this further? I'm super thankful that you answered so fast, that's amazing

ttlappalainen, 2024-09-30:
Pins 14 and 15 has been used for outputting PWM at boot and pin 15 is also strapping pin (see. https://randomnerdtutorials.com/esp32-pinout-reference-gpios/) I do not understand, why it would work differently in different USB. Anyway I have never used those pins for CAN. Normally I have 16 and 17.

MartinVerges, 2024-09-30:
I spend some time modifying it to be an ESP32 with MCP2515 using SPI as this is available as a secondary CAN port of the device. Unfortunately, exactly the same behavior and I can see all N2k frames on the Linux Serial but not on the Windows Serial. So I digged deeper and used `ArduinoGateway.ino` to send out periodic device information. It ends up with a output like 
```
...XNXNXNXNXNXNXNXNXNXNXNXNXNXNXNXNXNXN10326: Send PGN:126993
10326:  - can ID:502272271
60,EA,0,FF,FF,FF,FF,FF
10326 : Pri:7 PGN:126993 Source:15 Dest:255 Len:8 Data:60,EA,0,FF,FF,FF,FF,FF
XNXNXNXNXNXNXNXNXNXNXNXNXN...
```
so each "X" is a `FramesRead<MaxReadFramesOnParse` equals `true`. As it is now on the `tNMEA2000_mcp::CANGetFrame` there is no longer an `xQueueReceive` and I added the output of "N" if `CAN_MSGAVAIL == N2kCAN.checkReceive() ` fails.

[image]
And yes, I removed the R17 and R18 to ensure that this is not a problem.

I somehow begin to believe this panel PC is cursed and don't want to allow me to read the CAN bus. It's such a weird issue. My next and so far last step before giving up if no one else has an Idea is to switch the serial pins away from the USB port and use another serial to connect the esp32 output to the panel PC. Just have to search for some parts to make it possible.

MartinVerges, 2024-09-30:
I found a nother shield and esp32 here. This works and so I believe it's some strange hardware issue with this dual can PCB that I got. Sorry to bother you and thanks for your assistance! Keep up the good work, it's great stuff.

closed 2024-09-30 by MartinVerges