ttlappalainen/NMEA2000 #432: CAN_init shopuld be public

cosina, 2024-10-07:
Then the application can call CAN_Ini if, for whatever reason, communication can not be established upon starting EPS32. In my case ESP32 is tarted before power to NMEA was provided.

```cpp
  if ( updReinit.Update( 50000 ) ) {  // not to often
    unsigned int bus_status = MODULE_CAN->SR.B.BS;
    if ( bus_status == 1) {
       // bus off - reinit
      NMEA2000.CAN_init();
      TRACE( "Reinit bus" ); 
    }
  }
```

ttlappalainen, 2024-10-07:
No. That should be handled on higher level on tNMEA2000, since also it has to be restarted. I have that on my version, but have not published it.

phatpaul, 2024-10-18:
My driver is detecting when the CAN bus is down and then back up again.  It is successful in doing the bus recovery, but the library doesn't work after that without a restart.
https://github.com/phatpaul/NMEA2000_esp32xx/blob/b9a5c5e3ad5994b5cd3c55e160c9d8cd65bbbd7e/NMEA2000_esp32xx.cpp#L290

Couldn't the higher level library make use of this info to restart and redo the address claim etc?  So i.e. hot-plugging the bus doesn't work without adding my own application code?  Please advise on how best to do this.

ttlappalainen, 2024-10-19:
Email me directly so I can send you some instructions.