ttlappalainen/NMEA2000 #433: Some Library file missing

erkanrua, 2024-10-17:
NMEA2000\src/NMEA2000_CAN.h:317:10: fatal error: NMEA2000_mcp.h: No such file or directory
 #include <NMEA2000_mcp.h>
          ^~~~~~~~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: exit status 1

ttlappalainen, 2024-10-17:
As explained on document, you need to load also NMEA2000_mcp (https://github.com/ttlappalainen/NMEA2000_mcp) and CAN_BUS_shiel (https://github.com/ttlappalainen/CAN_BUS_Shield) libraries.

erkanrua, 2024-10-18:
Thank you for advice. After two repo installed solved my problem. Which document write it I didn't see. 

Thank you for your help

closed 2024-10-18 by erkanrua

ttlappalainen, 2024-10-18:
Did you read the library first page? On bottom it says "... full documentation ..."

[image]

Then there is heading "Get your project started" https://ttlappalainen.github.io/NMEA2000/get_started.html