ttlappalainen/NMEA2000 #435: Padding character in PGN 126996

thooge, 2024-11-06:
The padding in the fields modelId, softwareVersionCode, modelVersion and modelSerialCode in PGN 126996
ist set to 0xff. All these fields are of a static length of 256bit / 32 Byte.
I looked in de PGN-Description of canboat. That stated field type of ASCII-Text.
Also queried a "real world" Yacht Devices Temp Sensor which uses 0x20 as filler with is valid ASCII-Code.

The actual used padding 0xff is not valid ASCII.

So my question is: Is this a bug in the library or just incomplete specs?
I remember other issue with filler "@".

ttlappalainen, 2024-11-06:
126996: Unused characters shall be treated as not available, and filled with [0xFF] = Data not available.

@ is filler for AIS strings.

thooge, 2024-11-06:
Thanks for the information.
I will check out the behaviour of my other "real world" devices back on board.

ttlappalainen, 2024-11-06:
Check that the devices you compare has been certified. On the other hand padding with spaces is not totally wrong - then there is just spaces at end of the string. I have seen much more stupid solutions with other messages.