ttlappalainen/NMEA2000 #437: CAN TX Output not working on ESP32?

mattnj1975, 2024-12-10:
I have a Teensy 4.0 with a SN65HVD230 and the batterymonitor.ino sketch for testing, it works just fine onto my NMEA test backbone (terminated) when i put scope on the Teensy output pin > CAND on SN65HVD230 then i can see output from Teensy...as you would expect as it writes data...

Move to an ESP-WROOM-32, and there is no output (testing with scope)
If I load a receiving sketch then it works fine for receiving, when trying to transmit (again using standard batterymonitor.ino or TemperatureMonitor.ino) then nothing on the pins, I have tried using the default pins (16,4) and tested several other pin on ESP (my circuit was actually set to 0,4, and never any output.

I have tried 3 different ESP boards with totally different CAN chips, same output (nothing)

If i run a simple:-
digitalWrite(0, HIGH);
delay(2);
digitalWrite(0, LOW);
delay(2);

then I get a nice output on the pin (and NMEA bus) with scope.

Sketch runs fine with output in serial monitor:-
Interestingly if i disconnect the CAN chip, then i get warning...

PGN 130310 send failed
41745 : Pri:5 PGN:130310 Source:22 Dest:255 Len:8 Data:1,C1,70,FF,FF,FF,FF,FF
PGN 130311 send failed
42235 : Pri:5 PGN:130311 Source:22 Dest:255 Len:8 Data:1,C4,7D,73,FF,7F,FF,FF
PGN 130310 send failed
42245 : Pri:5 PGN:130310 Source:22 Dest:255 Len:8 Data:1,C1,70,FF,FF,FF,FF,FF
PGN 130312 send failed

When CAN chip is connected, I don't get the warning.

600725 : Pri:6 PGN:127508 Source:25 Dest:255 Len:8 Data:1,6B,5,33,0,B8,D,1
600735 : Pri:6 PGN:127506 Source:25 Dest:255 Len:11 Data:1,1,0,38,5C,82,2,C,0,FF,FF
600745 : Pri:6 PGN:127513 Source:25 Dest:255 Len:8 Data:1,D1,1,A4,1,35,7D,4B
602225 : Pri:6 PGN:127508 Source:25 Dest:255 Len:8 Data:1,6B,5,33,0,B8,D,1
602235 : Pri:6 PGN:127506 Source:25 Dest:255 Len:11 Data:1,1,0,38,5C,82,2,C,0,FF,FF
603725 : Pri:6 PGN:127508 Source:25 Dest:255 Len:8 Data:1,6B,5,33,0,B8,D,1

I have also just downloaded the latest NMEA2000 and NMEA2000_esp32 libraries.

[image]
[image]

mattnj1975, 2024-12-10:
i have answered my own question after a long evening was so simple:-

#define ESP32_CAN_TX_PIN GPIO_NUM_0 
#define ESP32_CAN_RX_PIN GPIO_NUM_4 
BEFORE the:-
#include <NMEA2000_esp32.h>

ttlappalainen, 2024-12-11:
Good you found it. 

Documenting is waste of time. Your problem is mentioned on
- https://github.com/ttlappalainen/NMEA2000_esp32/blob/master/NMEA2000_esp32.h
- https://github.com/ttlappalainen/NMEA2000/blob/master/src/NMEA2000_CAN.h
- https://ttlappalainen.github.io/NMEA2000/get_started.html under title "For use with ESP32"