ttlappalainen/NMEA2000 #438: Checks for pgn that is no longer active

mar0997, 2024-12-12:
Hi Timo - 

I continue to appreciate your work on the library and I have expanded my code significantly.  One thing that I have done is to create a new library based on your work that adds AC parameters 127744 - 127748 that I read from my Victron equipment.  I also added AC parameter 127503 that I then output to be read by my Czone equipment.  In addition to being able to now read the AC data by Czone I also add my 2 AC lines to get a total AC input to my boat.

This works great under normal situations when I am plugged into shore power or running my generator.  The issue that I have not been able to solve is that the AC data continues to show the last value when AC power is removed.  I have tried a variety of things to attempt to resolve this.  I assume it is something simple and it may or may not be related to the new library that I created.  

Using DataDisplay2 as an example I have tried this approach:

void ACVoltFreqB(const tN2kMsg &N2kMsg) {
  unsigned char SID;
  unsigned char ConnectionNumber;
  double ACRMSvoltageLtoN;
  double ACRMSvoltageLtoL;
  double ACFrequency;

  if (ParseN2kACVoltFreqB(N2kMsg, SID, ConnectionNumber, ACRMSvoltageLtoN, ACRMSvoltageLtoL, ACFrequency) )  {
    Line2Voltage = ACRMSvoltageLtoN;
    Line2Frequency = ACFrequency;
  }
  else {
    Line2Voltage = 0;
    Line2Frequency = 0;
  }
  }

I have also tried the approach of comparing the SID to the previous SID and if has not changed for a period of time to set the parameters to 0:

  ParseN2kACVoltFreqB(N2kMsg, SID, ConnectionNumber, ACRMSvoltageLtoN, ACRMSvoltageLtoL, ACFrequency);  {
    CurrentSID = SID;
    Line2Voltage = ACRMSvoltageLtoN;
    Line2Frequency = ACFrequency;
  
      if (CurrentSIDLastState != CurrentSID)
      {
        startAC2timer = millis();
        CurrentSIDLastState = CurrentSID;
      }

      if (millis() - startAC2timer >= 500 )
      {
        Line2Voltage = 0;
        Line2Frequency = 0;
      }
}

Both of these work fine when AC power is active.  But when AC power is removed I expect that the voltage and frequency go to 0, but instead both stay at the last values prior to removing power.  I have verified with my analyzer that the Victron equipment stops transmitting the pgn when I remove AC power so I'm stumped as to why neither of these work.  Can you offer me some advice?

ttlappalainen, 2024-12-12:
It is because your ACVoltFreqB will be never called after generator has been stopped and message will not be sent anymore. You have to do it like this:

```
static uint32_t AC2timer = 0;
static double Line2Voltage = 0;
static double Line2Frequency = 0;
#define ACTimeout 5000

void CheckTimeouts() {
  if ( millis()-AC2Timer > ACTimeout  ) {
    double Line2Voltage = 0;
    double Line2Frequency = 0;
  }
}

void ACVoltFreqB(const tN2kMsg &N2kMsg) {
  if ( ParseN2kACVoltFreqB(...) ) {
    ...
     AC2timer = millis();
  }
}

void loop() {
  NMEA2000.ParseMessages();
  CheckTimeouts();
  ...
}
```

mar0997, 2024-12-13:
Thank you Timo!  This makes sense to me.  I got tripped up on DataDisplay2.  With a small adjustment this works.  I really appreciate your help.