ttlappalainen/NMEA2000 #439: NMEA2000.EnableForward(false) not working for all output streams

dansonamission, 2024-12-16:
Hello

We have forwarding variables set like this

```
  NMEA2000.EnableForward(false);
  NMEA2000.SetForwardOwnMessages(false);
  NMEA2000.SetForwardSystemMessages(false);
  NMEA2000.SetForwardOnlyKnownMessages(false);
  NMEA2000.SetForwardStream(&Serial);
  NMEA2000.SetForwardType(tNMEA2000::fwdt_Text);
```

But we still see things like this on the serial port when the bus is down.

PGN 65299 send failed
PGN 127503, frame:0/4 send failed

Looking into the code it seem's ForwardStream is not being set to 0.

```
if (!result && ForwardStream!=0 && ForwardType==tNMEA2000::fwdt_Text) {
                ForwardStream->print(F("PGN ")); ForwardStream->print(N2kMsg.PGN);
                ForwardStream->print(F(", frame:")); ForwardStream->print(i); ForwardStream->print(F("/")); ForwardStream->print(frames);
                ForwardStream->println(F(" send failed"));
}
```

Have we got this wrong or is there another way to set ForwardStream to 0?

The read me indicates we can disable all forwarding, \ref tNMEA2000::EnableForward(false) to disable forwarding totally

Thanks

ttlappalainen, 2024-12-16:
You do not need to set forward stream at all, but it does not have anyway effect for those error messages.

dansonamission, 2024-12-16:
We're not setting ForwardStream, only the setforwarder's shown above.

We're trying to work out why ForwardStream is not equal to zero.

ttlappalainen, 2024-12-16:
Sorry my mistake. I remembered it wrong.

You have these settings:
```
  NMEA2000.SetForwardStream(&Serial);
  NMEA2000.SetForwardType(tNMEA2000::fwdt_Text);
```

Which overrides class constructor settings to 0.

```
tNMEA2000::tNMEA2000() {

#if !defined(N2K_NO_GROUP_FUNCTION_SUPPORT)
  pGroupFunctionHandlers=0;
#endif
#if !defined(N2K_NO_GROUP_FUNCTION_SUPPORT)
  InstallationDescriptionChanged=false;
#endif
  ForwardStream=0;
...
  ForwardType=fwdt_Actisense;
...
```

closed 2024-12-17 by dansonamission