ttlappalainen/NMEA2000 #442: DIY AutoPilot

h2omatt, 2024-12-22:
Timo, First off this library is such a huge help! 

I am trying to build an autopilot for my boat. I have the current NAC1 that integrates with the boat's MFD. Have you ever played with integrating your library with Simrad or other manufacturer MFDs to be used as an autopilot? right now I am looking for some advise on how figure this out. 

I was hoping if I created a device source with ClassCode: 40 and FunctionCode: 150. that the Simrad would pick it up as a Autopilot. 

[image]

I ran the Actisense Listener hooked up to the NAC-1 and got some more information but I'm concerned about all the manufacturer proprietary information. 

[image]

My test bench!

[image]

initially, I used NAC1 I ran the DeviceAnalyzer.ino to get a list of sources. But I don't see any of the PNGs coming from the Simrad.  equipment. I used Actisence listener to get some 

> 09:58:40.399 -> 
09:58:40.399 -> ----------------------------------------------------------------------
09:58:40.399 -> NAC-1   _Rudder feedback        
09:58:40.399 ->   Source: 20
09:58:40.399 ->   Manufacturer code:        275
09:58:40.399 ->   Unique number:            1712172
09:58:40.399 ->   Software version:         1.0.0.01                        
09:58:40.399 ->   Model version:                                            
09:58:40.399 ->   Manufacturer Information: 
09:58:40.399 ->   Installation description1: 
09:58:40.399 ->   Installation description2: 
09:58:40.399 -> 
09:58:40.439 -> 
09:58:40.439 -> **********************************************************************
09:58:40.439 -> ----------------------------------------------------------------------
09:58:40.439 -> NAC-1   _Autopilot              
09:58:40.439 ->   Source: 0
09:58:40.439 ->   Manufacturer code:        275
09:58:40.439 ->   Unique number:            1876012
09:58:40.439 ->   Software version:         1.0.0.01                        
09:58:40.439 ->   Model version:                                            
09:58:40.439 ->   Manufacturer Information: 
09:58:40.439 ->   Installation description1: 
09:58:40.439 ->   Installation description2: 
09:58:40.439 -> 
09:58:40.439 -> ----------------------------------------------------------------------
09:58:40.439 -> NAC-1   _Virtual rudder feedbac 
09:58:40.439 ->   Source: 1
09:58:40.439 ->   Manufacturer code:        275
09:58:40.439 ->   Unique number:            1744940
09:58:40.439 ->   Software version:         1.0.0.01                        
09:58:40.439 ->   Model version:                                            
09:58:40.439 ->   Manufacturer Information: 
09:58:40.439 ->   Installation description1: 
09:58:40.439 ->   Installation description2: 
09:58:40.439 -> 
09:58:40.439 -> ----------------------------------------------------------------------
09:58:40.439 -> NAC-1   _Rudder feedback        
09:58:40.439 ->   Source: 20
09:58:40.439 ->   Manufacturer code:        275
09:58:40.439 ->   Unique number:            1712172
09:58:40.439 ->   Software version:         1.0.0.01                        
09:58:40.439 ->   Model version:                                            
09:58:40.439 ->   Manufacturer Information: 
09:58:40.439 ->   Installation description1: 
09:58:40.439 ->   Installation description2: 
09:58:40.439 ->

ttlappalainen, 2024-12-22:
I have not tried to control NAC autopilots.

Your question is unclear. You said you are trying to make autopilot. What is problem with NAC-1? 

If you try to make your own, why you need to look NAC messages? Autopilots normally uses proprietary messages between display and autopilot unit. You can do the same with your display and autopilot unit. Autopilots often has some "navigation mode", where they listen PGN 129284 Navigation Data and PGN 129283 Cross Track Error from plotter system to correct course.

h2omatt, 2024-12-22:
Timo,

I am not trying to control the NAC-1?  Replace it with my own version.  I want to see if I can improve accuracy and I want to integrate some secondary inputs for remote controls and flybywire helms.  

The first challenge is make he Simrad recognize the Teensy as an available autopilot.  would you be able to help give me some guidance on that? 

[image]

ttlappalainen, 2024-12-23:
Not many ideas. Your unit should at least populate and sent PGN 127237 Heading/Track Control and PGN 127245 Rudder and you should also write PGN 126208 handlers - at lease request and command - for those. Then if some MFD supports general autopilot, it may see it.

Anyway I am afraid that Simrad as others uses proprietary messages to seek autopilot and they accept only their own. So if you connect Raymarine autopilot to the bus, Simrad would not regocnize it. Big companies are not much interested to support others systems since it would be out of their sales.