ttlappalainen/NMEA2000 #443: tN2kGNSStype incorrectly defined

lsoltero, 2025-01-04:
N2kTypes defines

enum tN2kGNSStype {
                            N2kGNSSt_GPS=0,                   ///< only GPS satellite network                                                                                            
                            N2kGNSSt_GLONASS=1,               ///< only the GLONASS satellite network                                                                                    
                            N2kGNSSt_GPSGLONASS=2,            ///< GPS + GLONASS satellite network                                                                                       
                            N2kGNSSt_GPSSBASWAAS=3,           ///< GPS + WAAS correction satellite network                                                                               
                            N2kGNSSt_GPSSBASWAASGLONASS=4,    ///< GPS + GLONASS + WAAS correction satellite network                                                                     
                            N2kGNSSt_Chayka=5,                ///< Russia's Chayka system is like the erstwhile LORAN-C radio navigation architecture                                    
                            N2kGNSSt_integrated=6,            ///< integrated mode                                                                                                       
                            N2kGNSSt_surveyed=7,              ///< surveyed mode                                                                                                         
                            N2kGNSSt_Galileo=8                ///< European Galileo satellite network                                                                                    
                          };

but these are off by one.  

0 is undefined not GPS.
1 is GPS.. etc
and
15 is missing which is used for manual position reports where the position is manual entered.

ttlappalainen, 2025-01-06:
Do yu mix tN2kGNSStype and tN2kGNSSmethod?

ttlappalainen, 2025-01-06:
Ah, someone added AIS messages has mixed tN2kGNSStype. That should have not been used in AIS messages and there should be on type.

lsoltero, 2025-01-06:
ah.. I think the issue really is that

struct tN2kAISAtoNReportData {

defined in N2kMessages.h defines

...

  /**********************************************************************//**
   * \brief Type of electronic position fixing device
   * See \ref tN2kGNSStype
  */
  tN2kGNSStype GNSSType;

where it really should be tN2kGNSSmethod.

lsoltero, 2025-01-06:
hm.. actually tN2kGNSSmethod does not work for the AIS types... and looking I see we have an issue with 129041, 129794

so I think a new type needs to be defined for the AIS GNS types.

I propose 
enum tN2kAISGNSSmethod {
                            N2kGNSSt_GPS=1,                   ///< only GPS satellite network                                                                                            
                            N2kGNSSt_GLONASS=2,               ///< only the GLONASS satellite network                                                                                    
                            N2kGNSSt_GPSGLONASS=3,            ///< GPS + GLONASS satellite network                                                                                       
                            N2kGNSSt_LORANC=4,           ///< Loran-C
                            N2kGNSSt_Chayka=5,                ///< Russia's Chayka system is like the erstwhile LORAN-C radio navigation architecture                                    
                            N2kGNSSt_integrated=6,            ///< integrated mode                                                                                                       
                            N2kGNSSt_surveyed=7,              ///< surveyed mode                                                                                                         
                            N2kGNSSt_Galileo=8                ///< European Galileo satellite network