ttlappalainen/NMEA2000 #449: PGN130074

Alyavin, 2025-04-20:
Hi Timi. Please advise. I want to get the coordinates of a point when saving it, this happens in pgn 130074. Is it possible to do this somehow with your library?

ttlappalainen, 2025-04-21:
It is possible but may be complex. Also depends, where do you set waypoint. If you do it in your chartplotter, it does not necessarily sent automatically 130074. Or it may not event support route and WP service PGNs 130064-130074. If it does support and send 130074 on setting WP, will it send single WP or whole list of your route WPs? If it does not sent them automatically, you need to request WPs. Unfortunately for request you need to know other information like database ID, route ID and WP ID. These information you get by requesting them with other PGNs.

Parsing 130074 is also not straightforward. It may contain n WP:s and one route may contain several 130074 messages.

Alyavin, 2025-04-21:
How can I get 130074 data for analysis? I couldn't find a parsing function in your library for 130074!

ttlappalainen, 2025-04-21:
Write it to your own code. I do not see to add parser to library due to reason I just explained above.

Alyavin, 2025-04-21:
Timi, can you help me, tell me how to write this correctly? I would like a fragment with this code!

ttlappalainen, 2025-04-21:
I am sorry but I too busy with other job. There are parser samples for other PGNs, so you can start there. To read all WP:s from single message you can use e.g., std::list to store WP information.

Alyavin, 2025-04-21:
ok/ Thank you