ttlappalainen/NMEA2000 #453: Floating point precision

stevehayles, 2025-05-24:
Firstly, what a great library,  it's a hugely valuable resource for lots of people. Thanks to @ttlappalainen and other contributors.  

I didn't see a live discussion section for this so I am posting here but without expecting the library authors to waste any time. I am worried about the performance of the library on something like an ESP32 having given that it largely parses values as double precision floats.  It feels like this level of precision often isn't needed and an ESP32 and other MCU's quite often handle 32 bit values in hardware and 64 bit values are software emulated.

GNSS position data is often an outlier but arguably float's in general aren't the right way to deal with coordinates anyway.  I am learning the codebase and was considering trying to fork a version that had either hard coded or configurable 32/64 bit values but I was keen to understand the practicalities of this before starting..

ttlappalainen, 2025-05-25:
I know commercial solutions using library with ESP32 and they are fine. Using just one type makes code much simple than starting to play with different types. If one is really experienced programmer and take everything out from MCU, one can improve performance by carefully handing different types. Many programmer will anyway create other bottlenecks. ESP32 can even handle converting data back and forth from double to json - I use ArduinoJson. It would be much more powerfull to use strict binary data structures for communication between ESP32 and browser and do all conversions with js.

MCU:s like ESP32 are used for small things like some sensor information handling. Companies who has made larger systems had used e.g., RPi. Either I do not see any sense to run library with smallest MCUs, which are on the performance limit and memory limit. Saving few bugs by using something smaller than ESP32 is only worth, if you sell thousands of them.

Look at https://acrossoceansystems.com/ product line. They have ESP32 as their heart. Devices can be configured and upgraded via WiFi (AP or client mode) or NMEA2000. You can connect them to your boat WiFi, read data with browser or use REST API. They have not used floats in code.

I would not waste more time than time of this message for thinking going down to float. I rather predict that 64-bit MCUs will also become more common in the future.

stevehayles, 2025-05-25:
Thanks for your prompt and considered reply @ttlappalainen, everything you say makes sense.  I will leave the decoding untouched and I can always cast to a float in any more critical code paths where the extra precision is not required.  Closing this (non) issue, thx

closed 2025-05-25 by stevehayles