ttlappalainen/NMEA2000 #454: How do you develop ESP32-based NMEA2000 solutions remotely — without access to the boat?

tomasenskede, 2025-06-26:
Hi all,

I'm working on a project involving an ESP32-based device that needs to communicate over an NMEA2000 network (SeaTalk NG compatible). However, like many of you, my boat is several hours away — which makes on-device testing slow and tricky.

I'm curious how others approach this development scenario. Do you:
- Set up a small bench network with a plotter or MFD (e.g. Raymarine Axiom, Garmin)?
- Use any simulation tools for sending/receiving NMEA2000 PGNs?
- Log raw data on the boat and replay it locally somehow?
- Rely on remote access to the onboard network?

Any tips, tools, or workflows that help you develop and test NMEA2000-based projects from home would be greatly appreciated. Thanks in advance for sharing your experiences!

ttlappalainen, 2025-06-26:
Have you checked my NMEA-Simulator https://kave.fi/Apps/

cram001, 2026-01-30:
The simulator is a great tool.  My final debugging is done with the code running on the boat, but monitored remotely. 

I use MikroTik routers on the boat. They have built-in support for WireGuard, which is like a private VPN. Many of their models can also run directly off DC 12V (wide voltage input).  One router is powered via POE from another to facilitate wiring. When I know I have code to troubleshoot, I`ll leave part of the N2K network powered ON. 

From home (or anywhere for that matter), I enable the WireGuard interface from my laptop (or phone or tablet). My computer then behaves just like I`m on the boat. I can put the IP address of any network device in the browser, I can OTA code directly to the ESP32, use my ethernet-N2K adapter to monitor traffic, etc....  If I want to see what is happening on the boat, I leave a webcam facing one of the Raymarine displays.

I use the Mirkotik Hap EX2.