ttlappalainen/NMEA2000 #457: Multiple NMEA2000 Instances

ocean-display, 2025-08-02:
Hello,

The program I am writing needs to receive NMEA2000 PGNs in one can bus interface and send them on another. I cannot find an example on how to do multiple instances of a tNMEA2000. I'd like to use NMEA2000_CAN.h so that it correctly configures the instances for me. The only idea I can come up with is to #define NMEA2000 to a different name, so that the name gets overridden when new creates the new instance. Is there another easier or more correct way?

Thanks,
Ernesto

ttlappalainen, 2025-08-03:
You need to create at least second instance manually. For first you can use default with NMEA2000_CAN.h or define both manually. If you use e.g., ESP32, you can use internal as default and then you need external MCP2525 controller with mcp.
```
#include <Arduino.h>
#define ESP32_CAN_TX_PIN GPIO_NUM_16
#define ESP32_CAN_RX_PIN GPIO_NUM_17
#include <NMEA2000_CAN.h>  // This will automatically choose right CAN library and create suitable NMEA2000 object
#include <N2kMessages.h>

// Define other instance manually
#include <SPI.h>
#include <mcp_can.h> // use one under https://github.com/ttlappalainen/CAN_BUS_Shield

#define N2k_SPI_CS_PIN 53
#define N2k_CAN_INT_PIN 21
#define MCP_CAN_CLOCK_SET MCP_16MHz
#define MCP_CAN_RX_BUFFER_SIZE 50

tNMEA2000 &NMEA2000_2=*(new tNMEA2000_mcp(N2k_SPI_CS_PIN,MCP_CAN_CLOCK_SET,N2k_CAN_INT_PIN,MCP_CAN_RX_BUFFER_SIZE));

```
Ten you need to do setup for both NMEA2000 and NMEA2000_2 instances.

ocean-display, 2025-08-08:
Hi,

I am using Teensy 4.1 for this.

I am opening creating the two instances as follows:

*(new tNMEA2000_Teensyx(
								tNMEA2000_Teensyx::CAN1,
								tNMEA2000_Teensyx::pinDefault,
								tNMEA2000_Teensyx::pinDefault ) );

*(new tNMEA2000_Teensyx(
								tNMEA2000_Teensyx::CAN2,
								tNMEA2000_Teensyx::pinDefault,
								tNMEA2000_Teensyx::pinDefault ) );

CAN1 works as expected with no issues. However, CAN2 is not. There seems to be traffic on the CAN bus, where there should be none. On the receiver side, I do not get any recognized packets. Only thing I can think of is that the TX pin is not being setup correctly. If I look at the TX pin with a scope, there is activity, which seems to be repeated in the RX pin. I believe this is the expected behavior. Am I doing this correctly using pinDefault?

Ernesto

ttlappalainen, 2025-08-09:
I am pretty sure that I have tested also with CAN2. I do not remember, have I ever tested with both at same time.

ocean-display, 2025-08-09:
Timo,

I am understanding this better. It appears that the instance in CAN2 ( I have shuffled them around trying to find a pattern) is trying to send the address claim packet when I "open" the instance. I have this canbus connected to another teensy (3.6) to receive the data. I can see the data being received by the second teensy on the Rx pin. I do not see anything in the Tx pin to serve as an ack of the message. I learned that the can controller will continue attempting to send the same message until it gets an ack. I do have the termination resistors.

I have the set the mode to tNMEA2000::N2km_ListenAndNode on both sender and receiver.

Not sure why it is not working correctly. I have used the Teensy 3.6 as a nmea2000 receiver, so I know the canbus works. I have these on pcb's, so I know I don't have loose connections.

Ernesto