ttlappalainen/NMEA2000 #461: 126998 should handle ConfigurationInformation for multiple devices

night199uk, 2025-08-28:
The implementation of ConfigurationInformation is currently global to NMEA2000, and not per-device-index like e.g. ProductInformation. There shoule probably be a ConfigurationInformation per iDev and the handler for e.g. 126998 should be handle multiple iDevs.

ttlappalainen, 2025-08-28:
Configuration information shows manufacturer data and information about installation.  If you have one physical device how it can be made by different manufacturer and how it can be installed to different place?

night199uk, 2025-08-28:
[image]
In the image, the "Name" field is mapped to InstallationDescription1 by Victron.